arpruss/gcodeplot #3: Pen Width

kyberkiller, 2017-03-02:
Nice tool but what parameter is defining the width of the pen?

arpruss, 2017-03-02:
None. Why do you need to define it?

kyberkiller, 2017-03-02:
Hi,

Thanks for the fast answer.

Maybe I have got the idea totally wrong? I thought that the idea of lib is
to produce g-code to draw the lines of svg?

And if you have a pen like 0.3 mm, a line for example 1.0 mm, surely it
matters how many lines are drawn next to each other to produce the original
line?

I'm actually trying to draw Kicad PCB SVG.

Thanks in Advance,

Tom

Tom Forsell
Linnustajantie 9
02940  ESPOO
tom.forsell@gmail.com
+358 (0)40 5859 067

arpruss, 2017-03-02:
Oh, that makes sense. Alas, gcodeplot doesn't support line thickness. It only draws a single line for each line in the svg. It's not a bad idea for the future, though, to add support for thickening lines. But to do that properly, I'd also have to do calculations regarding line-caps, and that would be annoying to do.

In the meanwhile, you can use long, narrow rectangles and shading, but that will result in way more pen movement, unless you're lucky enough that the rectangles align with the shading angle.

kyberkiller, 2017-03-04:
Here's a couple of my snippets I have used to parse gerber, just presenting the idea of an ortoghonal vectoring. parameter res is like pen width here:

var orto = getVector(prevObj.coord, obj.coord,res);
    
     var ortosteps = Math.trunc(Math.round(s.currenttool.params[0]/res)); 
     
     var xstart = prevObj.coord.x; var ystart = prevObj.coord.y;
     var xend = obj.coord.x; var yend = obj.coord.y;
     
     for ( var i = 1; i <= ortosteps;i+=1 ){  
       // todos: 
       // - draws only from beginning to end, moves back
       // - is not symmetric to center of line
       
       gcode += 'G1 X'+ xstart.toFixed(4) +' Y'+ ystart.toFixed(4) + ' F' + Fmove + "\n";
       //draw line
       gcode += 'G1 X'+ xend.toFixed(4) +' Y'+ yend.toFixed(4) + ' E' + Fext + ' F' + Fdraw + "\n";        
       xstart = xstart+orto.CW.stepX;
       xend = yend+orto.CW.stepX;
       ystart = ystart+orto.CW.stepY;
       yend = yend+orto.CW.stepY;
       if (ystart < 0 || yend < 0 )
         
            {
                console ('error in coordinates ' + JSON.stringify(prevObj) + "    " + JSON.stringify(obj));
            }
       
        }

 .... 
function getVector(start, end,res){   //returns orthogonal step vectors
          
    var dX = end.x-start.x;
    var dY = (end.y-start.y);
    var l = Math.sqrt(Math.pow(dX, 2)+Math.pow(dY, 2));
    var CW = {};var CCW = {};
    CW.stepX = res* (dY/l) ;
    CW.stepY = (-dX/l) * res;
    CCW.stepX = (-dY/l) * res;
    CCW.stepY = (dX/l) * res;
    return (JSON.parse(JSON.stringify({'CW':CW,'CCW':CCW})));
}

laserguru, 2017-06-26:
Kyberkiller: this program works with, for example, a vinyl cutter, where your "line" is infinitely (or at least very) thin. So if you want to cut a line on the vinyl, you cut both edges of the line, so the thickness is given by the distance between them

For processing a hatching pattern for line thicknesses, a lot can be drawn from slicer programs like Slic3r and their infill algorithms. Drawing an outline first, like when 3D printing, would probably be useful; then draw an "infill" hatch with the selected pattern and density %. Pen flow would probably be a factor since slower moves result in thicker lines, perhaps this could be controlled?    
If kyberkiller is drawing an SVG from a board design, I presume it's an automated way for using a marker as etch resist? If so, then a thick, even line is necessary so it effectively protects those areas of the copper.

tsipa, 2019-02-23:
+ for thickness feature. I'ts virtually impossible now to use ultra fine point marker for pcb.

nylen, 2019-10-14:
I think it's fair to say that making SVG strokes into solid objects (including end caps etc) is out of scope for this tool.

You might look at the "Path > Stroke to Path" option in Inkscape, which will convert a stroke into a filled object. Here's an example: https://graphicdesign.stackexchange.com/a/84571/144507

This program has features for filling/hatching, right? I'm not sure if that could be used here, but it might be better to do that in the SVG also, using something like the Inkscape extensions at https://github.com/evil-mad/EggBot/releases/tag/v2.8.1 / https://wiki.evilmadscientist.com/Installing_software.