arpruss/gcodeplot #4: Are you still developing this?

joeynuggetz, 2018-04-25:
Would be nice to specify our own pre and post commands for gcode. To make this more generic, I think perhaps the code out of the box should just zero out XYZ axis so all the user has to do is move the pen to the bed and hit print. Could you provide some input on what files would need to be changes to allow me to add pre and post gcode command instead of the ones currently in use?

arpruss, 2018-04-25:
gcodeplot.py

I am just too busy to do this right now.

The zeroing is a nice idea, though for some purposes a big blob of ink on the page or T-shirt or whatever would be annoying. So in practice I think one would need to put some plastic between the page and the pen there, and then remove it as the pen lifts.

qckbrnfx, 2018-05-04:
> So in practice I think one would need to put some plastic between the page and the pen there, and then remove it as the pen lifts.

This is exactly how I do it with the plotter I've been working on. I start with the pen at zero when I'm screwing it in to make sure it is making good contact with the paper. I just stick a thin piece of something under it while I'm doing these adjustments and then yank it out as soon as the pen lifts when I start a drawing.

suculent, 2018-06-06:
Well, I have also a bit of an issue getting the startpoint right. I was able to scale the workspace properly, but only when I skip the area which is an Y-offset of the pencil to actual printer head center. Would it help just to add the x and y-offset parameters? I could do it I guess.. I'm not a much of a python guy but I guess I could do it.

closed 2019-03-24 by arpruss