arpruss/gcodeplot #9: Lift/Down Commands

Arion02, 2019-01-04:
Hello.  Thank you for a great project.  I think the "Lift Command" and "Down Command" under the cutting settings tab for GCode creation is still under development?  It seems as though after every small movement, the down command is written in the gcode again.  Example below.  I have to include a statement such as "M3 S25 G4P0.3" to add a delay after the down command due to using a servo for the up/down movement.  Same thing for the lift command "G4P0.3  M5".  You can see the output will cause major delays when ran.  I am just trying to figure out a way to utilize the tool, but it may be beneficial to be able to add a delay before and after the actual cut path is started.  It is hard to find a tool that has the overcut and offset feature such as your tool, so thank you for your work!

G4P0.3  M5
G00 F2400.0 X5.716 Y16.641; move !!Xleft+5.716 Ybottom+16.641
M3 S25 G4P0.3
G01 F2100.0 X5.932 Y16.696; draw !!Xleft+5.932 Ybottom+16.696
M3 S25 G4P0.3
G01 F2100.0 X6.095 Y16.848; draw !!Xleft+6.095 Ybottom+16.848
M3 S25 G4P0.3
G01 F2100.0 X6.165 Y17.060; draw !!Xleft+6.165 Ybottom+17.060
M3 S25 G4P0.3
G01 F2100.0 X9.014 Y17.063; draw !!Xleft+9.014 Ybottom+17.063
M3 S25 G4P0.3
G01 F2100.0 X9.010 Y17.004; draw !!Xleft+9.010 Ybottom+17.004
M3 S25 G4P0.3
G01 F2100.0 X9.535 Y16.930; draw !!Xleft+9.535 Ybottom+16.930
M3 S25 G4P0.3
G01 F2100.0 X9.521 Y16.865; draw !!Xleft+9.521 Ybottom+16.865
M3 S25 G4P0.3

arpruss, 2019-01-04:
I just tried to commit a fix for this issue. Try it.

Arion02, 2019-01-05:
That is great! I will test it as soon as I can. I am in the middle of traveling.  Will report back here after I check it out.

Arion02, 2019-01-05:
Arpruss,

Looking at the gcode, it looks like your fix did the trick!  I won't be able to test on the vinyl cutter until a couple weeks from now, but everything looks good in the gcode.  Thank you for your prompt response and looking into my issue!  This is the only Inkscape plugin I could find that has the features you have implemented and supports gcode.  It is great!

Arion02, 2019-01-16:
@arpruss 

I have had a chance to test your changes.  I found a few issues so far:

1. It seems as though the length of the gcode commands are too long.  If I remove the comments, the gcode executes.
G00 F2400.0 X5.716 Y16.641 **; move !!Xleft+5.716 Ybottom+16.641**
the bold text I believe does something with GRBL to where it will not execute.

2. The initial move does not execute a down command, so it never actually lowers the cutting blade.  I can manually add it after the gcode is generated, but run into issue number 3.

3.  When moving between two cuts, the blade never raises, so the movements are also cut if the blade is down.

I tried to remove part of line 408 in gcodeplot.py to get rid of the comments, but I keep getting errors in the execution.  

I tried to fix it, but my programming skills are obviously lacking because I was unsuccessful!
Thanks.

arpruss, 2019-02-22:
Ad 1: You can add --comment-delimeters=none to your commandline. That should fix that.

arpruss, 2019-02-22:
I can't duplicate problems 2 and 3. I ran a test file with --lift-command=LIFT and --down-command=DOWN and I saw a DOWN at the beginning of the drawing and LIFT/DOWN around moves.

Arion02, 2019-02-22:
@arpruss Thank you for checking into this!

I am using a gcode sender utility, so I will have to figure out how to use the --comment-delimiters=none command you referenced in my utility.  I installed inkscape and your gcodeplot on another computer and it DOES give the desired output!  I am not sure what happened to the install on my old computer.  I even reinstalled inkscape and the gcodeplot files on the old computer and it did not resolve the issue.  Anyway, I am going to give this a shot on some vinyl this weekend.  Thank you again!

Arion

closed 2019-02-22 by Arion02