arpruss/gcodeplot #11: svg2cookiecutter.py broken after pull request changes

halfluck, 2019-03-22:
I'm unable to run svg2cookiecutter.py after the latest pull request, I get this error

Traceback (most recent call last):
  File "svg2cookiecutter.py", line 223, in <module>
    print(svgToCookieCutter(sys.argv[1]))
  File "svg2cookiecutter.py", line 171, in svgToCookieCutter
    line = Connector('connector'+pathName, points, fill)
  File "svg2cookiecutter.py", line 140, in __init__
    super().__init__(pathName, points, fill, False, None) # no stroke for connectors, thus no use of self.height and self.width
TypeError: super() takes at least 1 argument (0 given)

arpruss, 2019-03-22:
python 2.x compatibility issue. Fixed.

closed 2019-03-22 by arpruss

halfluck, 2019-03-23:
Working! The new features are AMAZING, so impressed right now.
Do you know if the inside walls has been improved to work?

arpruss, 2019-03-23:
What was the problem with inside walls?

halfluck, 2019-03-23:
> What was the problem with inside walls?

'holes' in feature elements within the perimeter of the cookie shape doesn't work with polygon fill
So for example the 'O' and the 'E' in one below

[image]

[image]