arpruss/gcodeplot #13: Setting align causes error

SanZamoyski, 2019-05-15:
```
python gcodeplot.py -c sggw.conf heart.svg > heart-test.gcode 
Traceback (most recent call last):
  File "gcodeplot.py", line 1129, in <module>
    plotter=plotter, gcodePause=gcodePause, pens=pens, pauseAtStart=pauseAtStart, simulation=svgSimulation)
  File "gcodeplot.py", line 363, in emitGcode
    scale.align(plotter, xyMin, xyMax, align)
  File "gcodeplot.py", line 144, in align
    o[i] = 0.5 * (self.xyMin[i] - self.scale[i]*xyMin[i] + self.xyMax[i] - self.scale[i]*xyMax[i])
AttributeError: 'Scale' object has no attribute 'xyMin'
```

when using config:

> init-code=;
> area=105,35,205,135
> no-allow-repeats
> gcode-pause=@pause
> scale=down
> align-x=none
> align-y=center
> tolerance=1.0
> no-send
> send-speed=115200
> input-dpi=1016,1016
> safe-delta-z=5
> lift-delta-z=2
> work-z=-10
> pen-down-speed=35
> pen-up-speed=40
> z-speed=5
> no-hpgl-out
> shading-threshold=1
> shading-lightest=3
> shading-darkest=0.5
> shading-angle=45
> no-shading-crosshatch
> no-stroke-all
> optimization-time=0
> sort
> no-pause-at-start
> extract-color=all
> tool-offset=0.000
> overcut=0.000
> no-simulation
> direction=none
> lift-command=none
> down-command=none
> comment-delimiters=;

>

SanZamoyski, 2019-05-15:
Also, setting 
```
scale=down-only
align-x=none
align-y=none
```
doesn't work too.

I'm not a python guy.

SanZamoyski, 2019-05-15:
In align, in Scale

```
            elif align[i] == ALIGN_CENTER:
                o[i] = 0.5 * (plotter.xyMin[i] - self.scale[i]*xyMin[i] + plotter.xyMax[i] - self.scale[i]*xyMax[i])
                #o[i] = 0.5 * (self.xyMin[i] - self.scale[i]*xyMin[i] + self.xyMax[i] - self.scale[i]*xyMax[i])
```

Seems to work.

Also in fit in Scale:
```
        self.offset = list(plotter.xyMin[i] - xyMin[i]*self.scale[i] for i in range(2))
        #self.offset = list(self.xyMin[i] - xyMin[i]*self.scale[i] for i in range(2))
```

should also, didn't check.

arpruss, 2019-05-15:
Fixed, I think. Try it.

closed 2019-05-15 by arpruss

SanZamoyski, 2019-05-16:
Yes, it works fine now out of the box.
Thanks!