arpruss/gcodeplot #14: Adding default config file

SanZamoyski, 2019-05-16:
Hi again!

Is there any way to put somewhere some default config file instead of using switches?

Regards!

arpruss, 2019-05-16:
No. But you can use --config=filename.

nylen, 2019-10-14:
You probably want to make a short shell script with your configuration.

Example:

```
#!/bin/bash

set -e

svg="$1"
gcode="${svg%.svg}.gcode"

python "$(dirname "$0")/gcodeplot/gcodeplot.py" \
    --work-z=0.9 \
    --lift-delta=0.6 \
    --safe-delta-z=3 \
    --z-speed=45 \
    --pens=pens.txt \
    "$svg" \
    > "$gcode.tmp"

# This avoids overwriting the gcode file if the script fails
mv "$gcode.tmp" "$gcode"

echo "Wrote gcode file: $gcode"
```

Save that as `convert.sh` and then run `bash convert.sh yourfile.svg`.