arpruss/gcodeplot #16: "Drawing out of range" with negative coordinates

TheNitek, 2019-09-03:
My plotter has (0, 0) at the center, so I want to be able to use negative X and Y values. So I set Left and Buttom to negative values. Unfortunately this gives my a "Drawing out of range" showing coordinates that are NOT.
I had a look at the code, but I couldn't find a obvious reason.

TheNitek, 2019-09-03:
I found the problem:
scale.offset = (plotter.xyMin[0],plotter.xyMin[1])
breaks
scale.scalePoint(point)
for negative values.

arpruss, 2019-09-04:
Does your svg file have negative coordinates?

TheNitek, 2019-09-04:
Yes, because my plottable area is -40 to 40 in both directions

arpruss, 2019-09-04:
Have you tried using positive coordinates and letting gcodeplot offset it to the negative area?

TheNitek, 2019-09-04:
No I didn't yet, because that doesn't seem to be available within the UI. Also I find it to be more intuitive to have the same coordinates in Inkscape as on my plotter.

arpruss, 2019-09-04:
That's I think the source of the problem: the plugin automatically adjust the drawing location so that the drawing's 0,0 corresponds to the minX,minY on the plotter.

nylen, 2019-10-14:
> That's I think the source of the problem: the plugin automatically adjust the drawing location so that the drawing's 0,0 corresponds to the minX,minY on the plotter.

I think this is a bug. This program should be looking at the min and max coordinates of the SVG, not assuming they are 0,0.