arpruss/gcodeplot #18: "Drawing out of range" - floating-point issues

nylen, 2019-10-14:
```
$ python gcodeplot/gcodeplot.py --work-z=0.9 --lift-delta=0.6 --safe-delta-z=3 --z-speed=45 --pens=pens.txt mandala.svg
Optimizing...[0%][5%][10%][15%][20%][25%][30%][35%][40%][45%][50%][55%][60%][65%][70%][75%][80%][85%][90%][95%]
Transport time improvement: 89.1% (took 1.64 seconds).
Drawing out of range: [0.0, 5.291666695939057e-06] [176.38888447916665, 176.38888977083334]
No points.
```

Files needed to reproduce the issue: [gcodeplot-fpbug.zip](https://github.com/arpruss/gcodeplot/files/3725947/gcodeplot-fpbug.zip)

I commented out the "Drawing out of range" section to make the print work, and it turned out awesome. I think it would be better to add a tolerance here though (or just allow negative values, since they're not necessarily invalid).

pooltable500, 2021-04-25:
Not sure if this is too late to ask, and sorry for bringing up an old post. I am having the same issue with having negative coordinates as I am using this on my 3018 CNC. Can you point out what lines you commented out?

arpruss, 2021-04-25:
I don't recommend this fix as you lose the safety measure: your cnc may bump against the edges. Instead, use the --area commandline argument.

sashokbg, 2021-11-28:
I have the same problem. Is there a solution besides commenting out the exception ?