arpruss/gcodeplot #23: ModuleNotFoundError: No module named 'svgpath'

amiee101, 2020-10-14:
I am new to Inkscape, and I installed the extension per the instructions, but I get this error, so the file cannot be saved as .scad: 
Traceback (most recent call last):
  File "svg2cookiecutter.py", line 2, in <module>
    import svgpath.parser as parser
ModuleNotFoundError: No module named 'svgpath'

Any ideas?

arpruss, 2020-10-14:
Did you copy the svgpath folder?

amiee101, 2020-10-15:
I thought so. I copied it from preferences - the user extensions part. So
it is in there. I even tried uninstalling and reinstalling the earlier
version of Inkscape along with the extensions.
It killed my brain for yesterday. I will be working on it (as I am sure its
user error) today.
Thank you!
Amiee

arpruss, 2020-10-15:
Also, note that this is not designed for Inkscape 1.x.

amiee101, 2020-10-15:
That could have been the issue for most of yesterday. Later on, I reverted
to Inkscape 0.92.5 (2060ec1f9f, 2020-04-08) and got the same error, so I
suspect the issue is with my lack of doing something on
the svgpath.parser front.

amiee101, 2020-10-16:
The issue was that I am, in fact, a moron.
Thanks for responding anyway :) Much appreciated, on both that front, and
for making the extension.

arpruss, 2020-10-16:
No worries!

closed 2020-10-16 by arpruss