arpruss/gcodeplot #25: SyntaxWarning: "is not" with a literal. Did you mean "!="?

LazerPlatypus, 2020-11-05:
The issue is found in parser.py and gcodeplot.py. using the find & replace tool to replace all occurrences of "is not" with "!=" seems to have resolved the problem, but I have not been able to thoroughly test that this is a good solution and that it does not introduce any more bugs. attached are my modified parser.py and gcodeplot.py files.
[gcodeplot_fix.zip](https://github.com/arpruss/gcodeplot/files/5497719/gcodeplot_fix.zip)

arpruss, 2020-11-05:
I'm usually pretty careful with these things and think about the difference, so if it's in my code, I wouldn't change it.

LazerPlatypus, 2020-11-05:
The only reason that I changed it, is because otherwise, the script won't run (it throws that error, and doesn't produce gcode). I would definitely go through and deliberately change what needs to be changed if it was my program, but I'm not familiar with the program, and rarely code in python.

arpruss, 2020-11-05:
But that's a warning, not an error.

LazerPlatypus, 2020-11-05:
I just undid my changes and reran the program. this is the warning screen that pops up:
[image]
the next bit is a little sporadic:
clicking 'OK' or 'x'-ing out of the window results in this error screen:
[image]
and doesn't produce any gcode.
sometimes, if you try saving it twice, it does work though, still displaying the warning screen, but the error screen doesn't pop up, and gcode is produced. I don't fully understand it.

arpruss, 2020-11-05:
Ah, I think that's not my code. :-) Try changing just that one is not to !=

LazerPlatypus, 2020-11-05:
doing that, you get the same warning, but this time in gcodeplot.py on line 456

arpruss, 2020-11-05:
Can you post your svg file?

LazerPlatypus, 2020-11-06:
[drawing-2.zip](https://github.com/arpruss/gcodeplot/files/5498309/drawing-2.zip)

closed 2020-11-16 by arpruss