arpruss/gcodeplot #26: [Bug] KeyError: 'd'

chepo92, 2020-11-15:
```
 File "svg2cookiecutter.py", line 232, in svgToCookieCutter
    for superpath in parser.getPathsFromSVGFile(filename)[0]:
  File "C:\Users\Axel\AppData\Roaming\inkscape\extensions\svgpath\parser.py", line 699, in getPathsFromSVGFile
    return getPathsFromSVG(ET.parse(filename).getroot())
  File "C:\Users\Axel\AppData\Roaming\inkscape\extensions\svgpath\parser.py", line 693, in getPathsFromSVG
    getPaths(paths, matrix, svg, path.SVGState(), {})
  File "C:\Users\Axel\AppData\Roaming\inkscape\extensions\svgpath\parser.py", line 587, in getPaths
    getPaths(paths, matrix, child, state, savedElements)
  File "C:\Users\Axel\AppData\Roaming\inkscape\extensions\svgpath\parser.py", line 587, in getPaths
    getPaths(paths, matrix, child, state, savedElements)
  File "C:\Users\Axel\AppData\Roaming\inkscape\extensions\svgpath\parser.py", line 549, in getPaths
    path = parse_path(tree.attrib['d'], matrix=matrix, svgState=state)
KeyError: 'd'
```
I Attach file to reproduce error 
[camion.zip](https://github.com/arpruss/gcodeplot/files/5543248/camion.zip)

arpruss, 2020-11-16:
I think the cause of the error is:
`<path style="fill:#000000;stroke:none;stroke-width:0.264583" id="path57" cx="51.671551" cy="96.268814" r="1.4382441" />`
in your svg.

This doesn't look like a valid path. I think it's meant to be a circle instead.

chepo92, 2020-11-16:
Thanks I'll review the file, at first I couldn't find which path was having trouble, i erased one by one and randomly appearing the error.
I added a try/except block to parser.py and it passed with no errors, may PR later, but first I'll identify the path

arpruss, 2020-11-16:
It's the one without d=.