arpruss/gcodeplot #34: Avoid turning loops?

Miq1, 2022-01-21:
Hi, 

first let me thank you for this convenient way to use my 3D printer as a cutter as well! 👍🏻 

I have a question on using it, though. Due to the nature of the blade's point running behind the turning axis, there are loops cut at sharp corners. Is there a way to avoid these? Something one could do in Inkscape to have the blade lifted, moved and lowered again at these corners?
[image]

arpruss, 2022-01-21:
The offset and overcut options may help with this.

Miq1, 2022-01-21:
Whoa! What a speedy reply! 😆 

Well, offset is obvious, 0.5mm in my case with 1mm shaft, 45 degree blades. But what does overcut do? How does the amount influence the tip trace?

Miq1, 2022-01-21:
I had the impression that the commercial plotters are doing a kind of "in place turn" when changing directions, i.e., moving the head in a circle section around the blade tip until it is pointing opposite to the next cut direction. So a tiny hole is cut at best, while maintaining the given paths very well.
Do you think this would be possible in GCodeplot as well?

arpruss, 2022-01-21:
It's been years since I've used this for cutting. I think overcut may be that kind of circle movement or some other solution to the same problem. I can't remember. I see that I used some code that I didn't write myself for the overcut, and it's pretty complicated code, so a quick glance at the code doesn't answer my question. I suggest the experimental method.  You might set the output of gcodeplot.py to HPGL so you can visualize the output.

Miq1, 2022-01-21:
I am a C++ programmer and have never done anything in Python, unfortunately. Else I would try myself understanding and in case modifying the code. 
Thanks anyway for your support!
Danke schön! 😉

arpruss, 2022-01-21:
Python is straightforward enough that one can do a lot of code modification without knowing python.

Miq1, 2022-01-22:
Uh, silly me! 😜 

When trying to understand the ``ProcessPath`` method I noticed that overcut actually is doing what I was suggesting! I was misled by the traces as can be seen in a GCode viewer, showing the strange loops at corners I thought were caused by the trailing blade.

But these traces in fact are drawn because overcut already does the corrections for the blade.

So excuse me for having raised a non-issue... 😊

closed 2022-01-22 by Miq1