arpruss/gcodeplot #35: Error when using greek characters

GhostGR, 2022-06-12:
When I use Greek characters i get the following error when I try to generate the gcode, if I remove the Greek characters it saves normally.

`Traceback (most recent call last):
  File "C:\Users\LT\AppData\Roaming\inkscape\extensions\mobi.omegacentauri.cookiecutter\gcodeplot.py", line 1073, in <module>
    data = f.read()
  File "C:\Program Files\Inkscape\lib\python3.9\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x9d in position 1640: character maps to <undefined>
`

arpruss, 2022-06-12:
You should convert text to strokes beforehand.

GhostGR, 2022-06-13:
I was doing object to path, i just tried stroke to path like you recommended and it works.

By the way thanks for the project, i use it to cut vinyl on the ender 3 and works great !

closed 2022-06-13 by GhostGR