arpruss/gcodeplot #36: Does not work on inkscape 1.1 - No points

alexiade, 2022-09-18:
It seems that upgrading breaks this plugin completely.  The object I have is definetly a path, but... alas, no points.

alexiade, 2022-09-18:
Managed to trace it down to this condition on stroking no longer working: path.svgState.stroke is not None . My path IS stroked, however it seems the state is no longer set. So nothing is exported.

alexiade, 2022-09-18:
small feature request too - could you add init-code and end-code to plugin definition?
Like so:

      <param name="init-code" type="string" _gui-text="Init Gcode" _gui-description="Pre-pended init commands"></param>
      <param name="end-code" type="string" _gui-text="End GCode" _gui-description="Appended end commands"></param>

I need them and I keep loosing them when I update the plugin...

arpruss, 2022-09-18:
Do you know how inkscape 1.1 marks strokes? Can you post an example of the svg that doesn't work?

alexiade, 2022-09-19:
Hmm. I tried to recreate it on windows with 1.2 and could not - path was properly exported. The bug fix list hinted that previous versions had a data loss issue... This may be caused by a bug in 1.1 that is the ubuntu LTS default, not in the plugin itself.