arpruss/gcodeplot #38: The called script passed additional data to Inkscape...

flow1990, 2022-11-21:
The called script passed additional data to Inkscape. No error message was returned from the script, but the result may be unusable.

If I try to save s.th. to gcode I get the above error message. Even if it says that there are no erros, if I click "ok" the next message says that it was not able to save the file.

Traceback (most recent call last):
  File "C:\Program Files\Inkscape\share\inkscape\extensions\gcodeplot\gcodeplot.py", line 9, in <module>
    import svgpath.parser as parser
  File "C:\Program Files\Inkscape\share\inkscape\extensions\gcodeplot\svgpath\__init__.py", line 1, in <module>
    from .path import Path, Line, Arc, CubicBezier, QuadraticBezier
  File "C:\Program Files\Inkscape\share\inkscape\extensions\gcodeplot\svgpath\path.py", line 3, in <module>
    from collections import MutableSequence
ImportError: cannot import name 'MutableSequence' from 'collections' (C:\Program Files\Inkscape\lib\python3.10\collections\__init__.py)

arpruss, 2022-11-21:
I think you're not using the latest version in the repository.

flow1990, 2022-11-21:
> I think you're not using the latest version in the repository.

I am using the latest one from 2021 or is there a newer one? It is version 0.11

arpruss, 2022-11-21:
Try 0.12 which I just posted. Or use an earlier Python.

The issue is Python 3.10 compatibility.

flow1990, 2022-11-22:
I don't get it, it is still the same... The only difference it that it now shows "No points." below the "The called script passed additional data to Inkscape. No error message was returned from the script, but the result may be unusable." but it is still not saveable.