arpruss/gcodeplot #39: Drawing gcode looks okay, but cutting gcode is bad

KuCZik, 2023-01-16:
Inkscape version 1.2.2
GCodePlot version 0.12
Viewing in CAMotics version 1.2.0

Same file, text is set to path in both scenarios, only change is between drawing/cutting in selection before saving. Is it bug, or am I doing something wrong? It is not problem of CAMotics, if I send it to cut on Klipper managed Ender 3 Pro, I get the same result.

Drawing GCode:
[image]

Cutting GCode:
[image]

arpruss, 2023-01-16:
That's the drag knife handling. Are you using a drag knife?

KuCZik, 2023-01-16:
Yes, this type 
[image]

Mounted like this
[image]

I have it in rubber band mount to have spring action. But the real result looks like this
[image]

And my config is
[image]

arpruss, 2023-01-16:
The default settings are for an angled blade whose cutting edge lags behind the center of movement -- a drag knife. You have a what looks like a bit with a sharp point centered on the center of movement. There should be settings under Cutting Settings to turn off the drag support. But I really know very little about cutting.

KuCZik, 2023-01-16:
No, I have a drag knife, inside is angled blade that is rotating, not a bit

arpruss, 2023-01-16:
Weird. My feeling is that it is not lifting up enough during the weird squiggles. Is there a parameter to adjust that in the Cutting Settings?

KuCZik, 2023-01-16:
In cutting settings I see just this
[image]

arpruss, 2023-01-16:
Does the Gcode include a lift for the loopy bits?

KuCZik, 2023-01-16:
This is the GCode (I just edited in gcodeplot.py start and end GCode, edited G00 and G01 commands to G0 and G1 - Klipper does not know what GOO and G01 is, and removed parking - made my own):

[gcodeplot.py.txt](https://github.com/arpruss/gcodeplot/files/10428463/gcodeplot.py.txt)
[test.gcode.txt](https://github.com/arpruss/gcodeplot/files/10428465/test.gcode.txt)

polhomarkho, 2023-05-21:
Hello, late to the party but wanted to add I had the same issue at first.  
The drag knife blade has an offset to the rotating axis (mine is 0.45mm) and the "tool offset" value in your config is probably too high at 1mm. You can lower it and with some trial and error you'll find the best value for you're drag knife.